richards-tech/RTIMULib #18: RTIMULib RTIMULib build on WIndows

uutzinger, 2015-01-05:
When I use CMake on windows for RTIMULib/RTIMULib and open the ALL_Build solution with VS2012 I get errors that it cannot find RTIMUSettings.h (in RTPressure.h) and RTMath.h (in RTIMU.h). Is there manual setting I am missing for the compiler? WIthout RTIMULib.lib I can not build RTIMUHost.

richardbarnett, 2015-01-05:
There was an omission in the RTHost CMakeLists.txt that has been fixed in the latest version. All you should need to do is to cmake using the RTHost CMakeLists.txt. It will build the library.

BTW there are also pre-built VS solutions in the RTHostIMU and RTHostIMUGL directories. These (especially the GL one) probably work better. The GL one does not pick up the texture images for some reason.

closed 2015-01-05 by richardbarnett

uutzinger, 2015-01-05:
Thank you very much.
The issues I had were:

-          CMAKE made nice VS solution for RTHost that compiled but did not link.

-          CMAKE made solution for RTLib but parser did not find all definition files.

-          The VS solution/project in the RTHost folder created a cmd.exe error 3 which might be an installation issue on my side. I had problem getting more verbose statement to fix this.
Thanks for your effort.