richards-tech/RTIMULib #19: cannot find -lRTIMULibGL and

luks3n, 2015-01-11:
the build script ("CMakeFile.txt" in subdirectory "Linux") seems to lag the compile script for the library "libRTIMULibGL", manually building this library was not successful either. Yet, building "RTIMULibDemoGL" IS possible, but fails to run with error "no gl capable x visuals available" and "segmentation fault", which is probably do to this missing library

richardbarnett, 2015-01-11:
Sounds like you are using an old Raspbian version. Please be more specific.

luks3n, 2015-01-11:
My version is "Linux version 3.12.35+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #733 PREEMPT Tue Jan 6 21:03:44 GMT 2015", just plain simple Raspian, updated/upgraded it just yesterday. 
When I do exactly what is written here https://github.com/richards-tech/RTIMULib/tree/master/Linux#build-using-cmake ,I get this error.

luks3n, 2015-01-11:
I just ignored this error and continued to follow the readme.md, everything else compiles without error, but after I type "RTIMULibDemoGL" in LXTerminal (obviously on  Desktop), I get
"QGLTempContext: No GL capable X visuals available."

richardbarnett, 2015-01-11:
Oh. That's a newer version of Raspbian than I have tested it with. Maybe something has changed.

richardbarnett, 2015-01-11:
One other thought. Try running qmake in the RTIMULibDemoGL directory and see if that's any better. That will help identify if its a qmake and/or cmake problem and, if it works, will at least get you going.

richardbarnett, 2015-01-11:
If you wanted to try something quickly, change line 49 of Linux/CmakeLists.txt from:

ADD_SUBDIRECTORY(${CMAKE_CURRENT_SOURCE_DIR}/RTIMULibGL)

to 

ADD_SUBDIRECTORY(${CMAKE_CURRENT_SOURCE_DIR}/RTIMULibGL RTIMULibGL)

and see if that helps.

luks3n, 2015-01-11:
Regarding qmake in RTIMULibDemoGL: it does compile just fine, without any error, does not change anything on the outcome.
Regarding change in Linux/CMakeLists.txt: I did change this, but does not have any effect either.

To me it seems like Linux/RTIMULibGL is not even touch by the build script, manually build this directory doesn't work either, after run cmake and make simply nothing happens.

richardbarnett, 2015-01-12:
Seems I need to find out what's going on - it seemed to be working correctly. 

Hopefully the qmake route will keep you progressing at least.

---------------------------------------
Richard Barnett
www.richards-tech.com

richardbarnett, 2015-01-19:
I believe this is now working.

closed 2015-01-19 by richardbarnett

peepo, 2015-01-24:
3.12.35+ #733 PREEMPT Tue Jan 6 21:03:44 GMT 2015 armv6l GNU/Linux

[100%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/RTIMULibDemoGL_automoc.cpp.o
Linking CXX executable RTIMULibDemoGL
/usr/bin/ld: cannot find -lRTIMULibGL
collect2: ld returned 1 exit status
RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/build.make:269: recipe for target 'RTIMULibDemoGL/RTIMULibDemoGL' failed
make[2]: *** [RTIMULibDemoGL/RTIMULibDemoGL] Error 1
CMakeFiles/Makefile2:375: recipe for target 'RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/all' failed
make[1]: *** [RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/all] Error 2
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2

using: https://github.com/richards-tech/RTIMULib/tree/master/Linux

fully up to date afaict

tried once more after editing line 49 same result

richardbarnett, 2015-01-24:
There definitely was a problem with this but I thought I had fixed it. If you had tried with an earlier version,  it might be worth deleting the build directory and starting again with that. I have noticed that sometimes history will linger unexpectedly.

I would think that the zip has the same information but you could just check the contents of the offending CMakeLists.txt and check that it's the same as the current one showing on GitHub.

peepo, 2015-01-24:
I have rm -r build a few times, had not used an earlier version
CMake seems identical

what else to do?

richardbarnett, 2015-01-24:
I am looking into it. Meanwhile, you could just use the qmake option to get going.

luks3n, 2015-01-24:
After trying the new version, I still get the same error. I removed everything and cloned the whole thing again. 

pi@raspberrypi ~/RTIMULib/Linux/build $ make
Scanning dependencies of target RTIMULib
[  2%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/RTFusion.cpp.o
[  4%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/RTFusionRTQF.cpp.o
[  7%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/RTMath.cpp.o
[  9%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/RTFusionKalman4.cpp.o
[ 11%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/RTIMUAccelCal.cpp.o
[ 14%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/RTIMUHal.cpp.o
[ 16%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/RTIMUMagCal.cpp.o
[ 19%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/RTIMUSettings.cpp.o
[ 21%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTIMU.cpp.o
[ 23%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTIMUGD20M303DLHC.cpp.o
[ 26%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTIMUGD20HM303DLHC.cpp.o
[ 28%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTIMUGD20HM303D.cpp.o
[ 30%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTIMULSM9DS0.cpp.o
[ 33%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTIMUMPU9150.cpp.o
[ 35%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTIMUMPU9250.cpp.o
[ 38%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTIMUNull.cpp.o
[ 40%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTPressure.cpp.o
[ 42%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTPressureBMP180.cpp.o
[ 45%] Building CXX object RTIMULib/CMakeFiles/RTIMULib.dir/IMUDrivers/RTPressureLPS25H.cpp.o
Linking CXX shared library libRTIMULib.so
[ 45%] Built target RTIMULib
Scanning dependencies of target RTIMULibDrive
[ 47%] Building CXX object RTIMULibDrive/CMakeFiles/RTIMULibDrive.dir/RTIMULibDrive.cpp.o
Linking CXX executable RTIMULibDrive
[ 47%] Built target RTIMULibDrive
Scanning dependencies of target RTIMULibDrive10
[ 50%] Building CXX object RTIMULibDrive10/CMakeFiles/RTIMULibDrive10.dir/RTIMULibDrive10.cpp.o
Linking CXX executable RTIMULibDrive10
[ 50%] Built target RTIMULibDrive10
Scanning dependencies of target RTIMULibCal
[ 52%] Building CXX object RTIMULibCal/CMakeFiles/RTIMULibCal.dir/RTIMULibCal.cpp.o
Linking CXX executable RTIMULibCal
[ 52%] Built target RTIMULibCal
Scanning dependencies of target RTIMULibDemo_automoc
[ 54%] Automoc for target RTIMULibDemo
Generating moc_AccelCalDlg.cpp
Generating moc_IMUThread.cpp
Generating moc_MagCalDlg.cpp
Generating moc_RTIMULibDemo.cpp
Generating moc_SelectFusionDlg.cpp
Generating moc_SelectIMUDlg.cpp
[ 54%] Built target RTIMULibDemo_automoc
[ 57%] Generating ui_RTIMULibDemo.h
Scanning dependencies of target RTIMULibDemo
[ 59%] Building CXX object RTIMULibDemo/CMakeFiles/RTIMULibDemo.dir/AccelCalDlg.cpp.o
[ 61%] Building CXX object RTIMULibDemo/CMakeFiles/RTIMULibDemo.dir/RTIMULibDemo.cpp.o
[ 64%] Building CXX object RTIMULibDemo/CMakeFiles/RTIMULibDemo.dir/IMUThread.cpp.o
[ 66%] Building CXX object RTIMULibDemo/CMakeFiles/RTIMULibDemo.dir/MagCalDlg.cpp.o
[ 69%] Building CXX object RTIMULibDemo/CMakeFiles/RTIMULibDemo.dir/SelectIMUDlg.cpp.o
[ 71%] Building CXX object RTIMULibDemo/CMakeFiles/RTIMULibDemo.dir/SelectFusionDlg.cpp.o
[ 73%] Building CXX object RTIMULibDemo/CMakeFiles/RTIMULibDemo.dir/main.cpp.o
[ 76%] Building CXX object RTIMULibDemo/CMakeFiles/RTIMULibDemo.dir/RTIMULibDemo_automoc.cpp.o
Linking CXX executable RTIMULibDemo
[ 76%] Built target RTIMULibDemo
Scanning dependencies of target RTIMULibDemoGL_automoc
[ 78%] Automoc for target RTIMULibDemoGL
Generating moc_AccelCalDlg.cpp
Generating moc_IMUThread.cpp
Generating moc_MagCalDlg.cpp
Generating moc_RTIMULibDemoGL.cpp
Generating moc_SelectFusionDlg.cpp
Generating moc_SelectIMUDlg.cpp
[ 78%] Built target RTIMULibDemoGL_automoc
[ 80%] Generating ui_RTIMULibDemoGL.h
Scanning dependencies of target RTIMULibDemoGL
[ 83%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/AccelCalDlg.cpp.o
[ 85%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/RTIMULibDemoGL.cpp.o
[ 88%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/IMUThread.cpp.o
[ 90%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/MagCalDlg.cpp.o
[ 92%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/SelectIMUDlg.cpp.o
[ 95%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/SelectFusionDlg.cpp.o
[ 97%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/main.cpp.o
[100%] Building CXX object RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/RTIMULibDemoGL_automoc.cpp.o
Linking CXX executable RTIMULibDemoGL
/usr/bin/ld: cannot find -lRTIMULibGL
collect2: ld returned 1 exit status
RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/build.make:269: recipe for target 'RTIMULibDemoGL/RTIMULibDemoGL' failed
make[2]: *** [RTIMULibDemoGL/RTIMULibDemoGL] Error 1
CMakeFiles/Makefile2:375: recipe for target 'RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/all' failed
make[1]: *** [RTIMULibDemoGL/CMakeFiles/RTIMULibDemoGL.dir/all] Error 2
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2

richardbarnett, 2015-01-24:
Hopefully latest version now has this fixed! I was being fooled by an earlier installation of the lib file masking this CMakeLists.txt problem on the Pi.

peepo, 2015-01-24:
well builds okay and make install, one can rum programs from their build directory,
but if one tries to run the installed version:
RTIMULibCal: error while loading shared libraries: libRTIMULib.so: cannot open shared object file: No such file or directory

richardbarnett, 2015-01-24:
Did you do the sudo ldconfig after sudo make install?

peepo, 2015-01-24:
ah yes... are you expecting to be able to use ssh?
I am:
 RTIMULibDemoGL 
QGLTempContext: No GL capable X visuals available.
Settings file RTIMULib.ini loaded
Using fusion algorithm RTQF
min/max compass calibration not in use
Ellipsoid compass calibration not in use
Accel calibration not in use
Segmentation fault

luks3n, 2015-01-24:
Compiling works perfectly now, thanks Richard!
@peepo I guess you are connecting via XRDP aka remote desktop? That's a common error, OpenGL does not work in XRDP, if you really want to, you can activate X11 forwarding on your SSH Client, and then setup a X Server on you desktop, you should be able to see something then. But it is REALLY slow this way.

peepo, 2015-01-24:
yes, but there are other ways to get openGL over ssh...
maybe not with a desktop?