richards-tech/RTIMULib #21: QGLTempContext: No GL capable X visuals available.

peepo, 2015-01-25:
originally I had thought this was an ssh issue, however I have now plugged a keyboard directly into the RPi, same result.

copied to: http://www.raspberrypi.org/forums/viewtopic.php?f=68&t=97658

from the web found: check the X11 configuration and your video driver. 
but not clear what I might do...

RTIMULibDemoGL
QGLTempContext: No GL capable X visuals available.
Settings file RTIMULib.ini loaded
Using fusion algorithm RTQF
Using min/max compass calibration
Ellipsoid compass calibration not in use
Using accel calibration
Segmentation fault

richardbarnett, 2015-01-25:
Are you running the desktop or just in command line mode? The desktop has to be running for any GUI app.

peepo, 2015-01-25:
startx, first then open terminal and run in terminal.

richardbarnett, 2015-01-25:
What version (build date) of Raspbian are you using? Qt OpenGL only started working at something like the September 2014 build I think.

peepo, 2015-01-25:
uname -a
Linux pi 3.18.3+ #741 PREEMPT Fri Jan 23 13:32:52 GMT 2015 armv6l GNU/Linux

after I ran rpi-update now:
i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

ls /dev
confirms no i2c at all ???

richardbarnett, 2015-01-25:
Did you set up the i2c as described here - https://github.com/richards-tech/RTIMULib/tree/master/Linux#enabling-and-configuring-the-i2c-bus ?

peepo, 2015-01-25:
yup, ran through that checking none of the documents had changed.

richardbarnett, 2015-01-25:
All I can say is that I have never seen that before! You might want to run:

dmesg | grep i2c

in case an error occurred.

peepo, 2015-01-25:
i2c /dev entries driver
was only listing

trying update, upgrade just in case...

peepo, 2015-01-25:
uname -a
Linux pi 3.18.3+ #741 PREEMPT Fri Jan 23 13:32:52 GMT 2015 armv6l GNU/Linux

now window opens very briefly, but back to:

RTIMULibDemoGL 
QGLTempContext: No GL capable X visuals available.
Settings file RTIMULib.ini loaded
Using fusion algorithm RTQF
Using min/max compass calibration
Ellipsoid compass calibration not in use
Using accel calibration
Segmentation fault

peepo, 2015-01-26:
Richard, 
is it correct that RTIMULib uses Qt4,
whereas current RPi uses Qt5?**
Does that indicate solution?

Jonathan

**ghans from RPi writes:

Qt insists on GLX support for some strange reason.
Qt5 should be aware of the Pis specific problems with GLX
and use OpenGLES + EGL instead automatically , but this
program seems to use Qt4 ?!

http://www.raspberrypi.org/forums/viewtopic.php?f=68&t=97688&p

I had wondered whether I might have leapfrogged updates after i2c experience yesterday...

richardbarnett, 2015-01-26:
I wasn't aware that Qt5 was available on the Pi without compiling it yourself. What I do know is this (since I did this the other day):

1. Get the latest version of Raspbian
2. Generate a new SD card with the latest version
3. Enable i2c
4. sudo apt-get install libqt4-dev cmake git
5.  git clone the repo
6. Build using cmake

That's what I did and it worked fine. I have never used rpi-update to change to a new version so cannot confirm if this is the same as producing a new SD card image with the latest version.

peepo, 2015-01-26:
what I believe is that anyone updating will get Qt5 as I did, 
so you may want to amend your instructions?

richardbarnett, 2015-01-26:
Enter:

qmake -v

at a command prompt. What version of Qt does it report?

peepo, 2015-01-26:
many thanks for your patience... I'm at a loss:
QMake version 2.01a
Using Qt version 4.8.2 in /usr/lib/arm-linux-gnueabihf

richardbarnett, 2015-01-29:
Have you tried starting with a clean install of the latest Raspbian? I am pretty sure that would work.

closed 2015-03-22 by richardbarnett