richards-tech/RTIMULib #22: "Overflow" of sensor fusion resulting in extremly strange behavior

luks3n, 2015-01-27:
I've changed the sampling rate from standard 80 Hz to 160 Hz, which resulted in very strange bahavior, namely completely off euler angles, which change absolut value and sign every period. Interestingly, the same config ran perfectly fine in RTIMULibDemo, so after some trial and error, I figured out that if I wait to initialize the imu (MPU9250) until right before my main loop, and immediately poll data without waiting, it returns to operating normally. But waiting just 100 ms (to ensure everything is setup correctly) BEFORE the loop, which is polling data, completely destroys euler angles. I guess some buffer overflows in this short time, which shouldn't be possible.

I use a  MPU9250 connect to Raspberry Pi via SPI (I2C broke when soldering pull-up resistor, but I have a new one for backup lying around).

richardbarnett, 2015-01-28:
Well 100mS is 16 samples...it's an interesting point. I'll investigate.

richardbarnett, 2015-02-01:
I can only assume that it is a FIFO overflow issue. The MPU-9250 only has a 512 byte FIFO (rather than 1024 as in the MPU-9150). At 160 samples per second, this will overflow in roughly 256mS. Probably then the frame boundaries get corrupted leading to incorrect data processing.

Certainly as soon as the MPU-9250 is initialized, it will start filling the FIFO. As to why 100mS caused the problem, I don't know. The code does attempt to recover from FIFO overflows but it is actually quite difficult to do properly. Certainly this is one disadvantage of it not being interrupt driven - it is up to the user code to poll frequently enough to avoid overflow. This isn't easy in a non real-time environment however.

richardbarnett, 2015-02-01:
Just discovered that the MPU-9250 driver assumes that the FIFO size is still 1024 so the FIFO overflow recovery code never operates. I am going to fix that...

closed 2015-03-22 by richardbarnett