richards-tech/RTIMULib #24: m_compassAdjust

uutzinger, 2015-02-20:
In RTIMUMPU9150.cpp or 9250.cpp the compass sensitivity values are applied after the x-y compass axes are flipped. 

I would assume the manufacturer stored compass sensitivity data with corresponding device axes and the axes should be flipped after applying the sensitivity correction.

In the Arduino version of the code, the flipping of the axes is done after the sensitivity adjustment. 
In the Teensy version, the flipping occurs like in this version. 

Can you please either correct the Arduino or this/Teensy version? 
Thank you.

richardbarnett, 2015-02-20:
Oh. Yes, it should be done before the flip. Thanks - I'll update that.

richardbarnett, 2015-02-21:
Fixed now.

closed 2015-02-21 by richardbarnett