richards-tech/RTIMULib #25: Feature request: Humidity sensors?

XECDesign, 2015-02-22:
Hello Richard,

Would adding support for humidity devices (namely HTS221) be outside of the scope for RTIMULib? If it's alright, I don't mind doing the work myself and sending the pull request when it's done.

richardbarnett, 2015-02-22:
The RTIMU_DATA structure does have a spot for humidity readings. Wasn't sure if it would ever be used though! It could be implemented a bit like how the pressure sensors are handled in that an extra call is made to fill in the humidity data in the RTIMU_DATA structure.

in reality though, this will fit in better with RTembedded but that is a not ready for primetime yet...

XECDesign, 2015-02-22:
Alright, thank you. I'll see how far I get with it.

closed 2015-02-22 by XECDesign