richards-tech/RTIMULib #28: undefined reference MS5611

luks3n, 2015-03-30:
Found a bug in newest version :
objects/RTPressure.o: In function `RTPressure::createPressure(RTIMUSettings*)':
RTPressure.cpp:(.text+0xe4): undefined reference to `RTPressureMS5611::RTPressureMS5611(RTIMUSettings*)'
collect2: ld returned 1 exit status

I am using a MPU9250 on SPI. The pressure sensor shouldn't cause any problems since the pressure sensor shouldn't work when connected with SPI anyway, right? Probably has something to do with the March 21 release, where support for MS5611 was added.

richardbarnett, 2015-03-30:
Which way are you building things? Makefile, CMakeLists...? In fact, what are you trying to build?

luks3n, 2015-03-30:
I build via makefile on Raspberry Pi, but it's one of my own projects, which uses a configuration file generated by RTIMULibCal (was generated before the update). Up until I updated today it compiled just fine, now this happens.

richardbarnett, 2015-03-30:
You probably need to add RTPressureMS5611.cpp to your Makefile then. It needs to compiled and linked in whether you eventually use it or not.

luks3n, 2015-03-30:
True, I thought I autocompiled everything that is in the RTIMULib directory, guess it wasn't. Solved.

closed 2015-03-30 by luks3n