richards-tech/RTIMULib #32: BNO055 + Raspberry pi + I2C = Fail

glabmoris, 2015-05-21:
I am experiencing the same unreliability problems reading from a bno055 through I2C when coupling it to a raspberry pi 2 B. I have noticed from your changelog that "This is being investigated". I'd be interested to hear of your findings.

Thanks !

glabmoris, 2015-05-21:
Ah I found the related post on your website...sad bug....too bad I wasted hours trying to find something wrong in my setup only to find it to be a well known hardware bug.

Thanks for documenting it, I'll close this issue :)

richardbarnett, 2015-05-21:
We can commiserate together - I also wasted ages on this! I wasn't familiar with the Pi problem and didn't  suspect that until I looked at the scope trace and saw some very nasty things going on.

glabmoris, 2015-05-21:
I have found a working workaround to getting the BNO055 to work. All you need to do is slow down the I2C bus, which is easier said than done with the new device files. However, the RPi engineers have provided an overlay that allows one to slow it down properly

For instructions on how to slow down the bus in the RPi 2 B:

https://www.raspberrypi.org/forums/viewtopic.php?t=97314

richardbarnett, 2015-05-21:
Nice work! However, since the chip does support a UART interface that runs at 115200, that would seem to be a better option given how slow you have to run the I2C interface.

glabmoris, 2015-05-21:
Indeed, however in my case the UART is already hogged by the GPS and I didn't feel like playing with IO extenders to get around it. But yes, the UART works very well too.

perezbecker, 2015-06-27:
@glabmoris, I am trying to hook up a BNO055 (adafruit breakout) to a raspberry pi, and am just seeing that this is not trivial. Do you mind sharing your code to save me some work? Any help would be appreciated!