richards-tech/RTIMULib #33: Vertical 9150 orientation and noisy measurements

timetogo2, 2015-06-03:
Hi,
I am using a 9150 positioned in a verticle mounting direction with pin 1 in the upper left corner with the writing on the chip so I can read it.

I am using RTIMU_XEAST_YUP and I believe that has the Z axis pointed toward me if I read the InvenSense and RTIMULib documentation correctly.  I am using the default slerp power value of 0.02.

I have observed the following general behaviors while running RTIMULibDrive on a Raspberry PI 2 connected to an Invensense 9150 evaluation board:

1) If I disable the compass I see drift accumulating in the orientation values.  I think that is normal for this chip and one of the reasons for fusion.

2) If I enable the compass I see a noisy yaw that varies substantially in the tenths place while roll and pitch vary in the hundredths place (10x).  This causes more than a degree of overall yaw noise which seems high to me.

3) If the magnetometer is not calibrated then the yaw will measure a new yaw orientation but then be pulled back toward some nominal value. I assumed this was because the compass is not calibrated but enabled and the fusion is using the uncalibrated compass values.  When I calibrate the magnetometer using RTIMULibCal then the yaw measures correctly, though still just as noisy.

4) If I disable the accelerations then the sign changes for roll, pitch, and yaw.  I do not see the sign change if I use the default axis rotation of RTIMU_XNORTH_YEAST and enable/disable the accelerations.  I am concerned that this sign change may be related to my noisy measurements.  The sign change is independent of whether the compass is enabled or not and dependent only on accelerations being enabled/disabled.

So what I have is either drift or noisy yaw and I was wondering if that is due to using RTIMU_XEAST_YUP and perhaps the sign changes I see when accelerations are included/excluded into the fusion.  Or maybe I don't have the axis rotation set correctly.  Or maybe there is something else I am not understanding.

I have read about some initial startup time that the IMU needs to be motionless to determine the biases for correcting drift.  Is that on power up? After resetting the IMU in SW?  Or is that something RTIMULib does on start up?

I truly appreciate the functionality RTIMULib provides but I just can't quite get it tuned in the way I would like it to be.  I even tried going through all 24 axis orientations to see if any worked better than the one I thought I should use (RTIMU_XEAST_YUP).

In summary there are two main questions:
1) Can I get less noise (or drift) in my configuration?
2) What is the proper drift bias startup procedure?

Thanks for any feedback/comments,
Eric

richardbarnett, 2015-06-03:
Eric,

Comments inline…

On Jun 3, 2015, at 3:10 PM, timetogo2 <notifications@github.com> wrote:

Hi,
I am using a 9150 positioned in a verticle mounting direction with pin 1 in the upper left corner with the writing on the chip so I can read it.

I am using RTIMU_XEAST_YUP and I believe that has the Z axis pointed toward me if I read the InvenSense and RTIMULib documentation correctly. I am using the default slerp power value of 0.02.

*RTIMULib has to do some axis swapping even in the default configuration so their documentation is not helpful. The best technique is this. Set the code back to the default RTIMU_XNORTH_YEAST. Then, orientate the IMU so that roll, pitch and yaw are all approximately 0. At this moment, the x axis points towards the north, y to the east and z straight down. Then move the IMU to the orientation you want, noting where the x, y and z axes now point. Use the appropriate setting that corresponds to this. To test, implement this setting and point the installed IMU towards the north and on a horizontal plane. Roll pitch and yaw should now be approximately 0.*

I have observed the following general behaviors while running RTIMULibDrive on a Raspberry PI 2 connected to an Invensense 9150 evaluation board:

1) If I disable the compass I see drift accumulating in the orientation values. I think that is normal for this chip and one of the reasons for fusion.

*This is residual gyro bias.*

2) If I enable the compass I see a noisy yaw that varies substantially in the tenths place while roll and pitch vary in the hundredths place (10x). This causes more than a degree of overall yaw noise which seems high to me.

*The MPU-9150 magnetometers are not the best. *

3) If the magnetometer is not calibrated then the yaw will measure a new yaw orientation but then be pulled back toward some nominal value. I assumed this was because the compass is not calibrated but enabled and the fusion is using the uncalibrated compass values. When I calibrate the magnetometer using RTIMULibCal then the yaw measures correctly, though still just as noisy.

*You are correct in your analysis here.*

4) If I disable the accelerations then the sign changes for roll, pitch, and yaw. I do not see the sign change if I use the default axis rotation of RTIMU_XNORTH_YEAST and enable/disable the accelerations. I am concerned that this sign change may be related to my noisy measurements.

*That’s odd. Not sure why that would be. Do you plan on running with the accelerometers off or is this just a test? Should not be relevant to noise.*

The sign change is independent of whether the compass is enabled or not and dependent only on accelerations being enabled/disabled.

So what I have is either drift or noisy yaw and I was wondering if that is due to using RTIMU_XEAST_YUP and perhaps the sign changes I see when accelerations are included/excluded into the fusion. Or maybe I don't have the axis rotation set correctly. Or maybe there is something else I am not understanding.

I have read about some initial startup time that the IMU needs to be motionless to determine the biases for correcting drift. Is that on power up? After resetting the IMU in SW? Or is that something RTIMULib does on start up?

*It accumulates 5 seconds worth of gyro bias but only when the accelerations and gyro rates are below a certain level. If you keep it still for 5 seconds after starting the library, you should be good to go but it isn’t essential as long as there are occasional times when it is not moving.*

I truly appreciate the functionality RTIMULib provides but I just can't quite get it tuned in the way I would like it to be. I even tried going through all 24 axis orientations to see if any worked better than the one I thought I should use (RTIMU_XEAST_YUP).

In summary there are two main questions:
1) Can I get less noise (or drift) in my configuration?

*Did I mention that the MPU-9150 magnetometers are horrible? If you look in RTIMU.cpp, there’s a parameter COMPASS_ALPHA. Making this value smaller (but not 0) smooths the magnetometer outputs more. You could also reduce Slerp but that will also reduce the accelerometer input. Then again, that may be ok. Try a Slerp of 0.01 or 0.005 and see if that helps.*

2) What is the proper drift bias startup procedure?

*As I said, best is to keep it still for 5 seconds if possible.*

Thanks for any feedback/comments,
Eric

timetogo2, 2015-06-04:
Richard,
I went through the procedure, got all the axes reporting near 0, and the axis rotation I end up with after re-positioning the IMU back to its vertically mounted position is RTIMU_XEAST_YDOWN.  Roll and pitch work perfectly but yaw says 17x degress and when I yaw the IMU the yaw value changes but fuses back towards 17x degrees within a couple of seconds.  I have gone through this procedure several times and with someone else watching just to make sure.

I then tried all 24 axis rotations by changing the RTIMULib.ini file and rerunning RTIMULibDrive each time.  None of the orientations would properly track the yaw and the orientations where the yaw reflected some yaw motion all seemed to track back to some nominal value.

I changed RTIMULibDrive to output the compass.x[yz]() values and they are indeed changing with the yaw motion.  I have tried two different 9150 evaluation boards so I don't think it is the 9150 itself.

Any more ideas?
Thanks,
Eric

richardbarnett, 2015-06-04:
I just tried the same set up - Pi, MPU-9150, RTIMU_XEAST_YDOWN with a default .ini file (i.e. no calibration data). It did what you are seeing - the yaw seems to revert to roughly a consistent value. I then ran RTIMULibCal in the same directory and calibrated the magnetometers. It then worked fine.

There is an important sequence here. You need to run RTIMULibDrive (or RTIMULibCal) to generate a default .ini file and then exit the program. Then, edit the .ini to select the axis rotation. Then run RTIMULibCal to calibrate the magnetometers. Then run RTIMULibDrive in the same directory as you ran RTIMULibCal (or else move the .ini to the new directory) and run RTIMULibDrive. This should work correctly. At least it does here!

Most important point - you must run the calibration after changing the axis orientation.

richardbarnett, 2015-06-04:
Additional thought - you must run the calibration with the IMU installed in whatever it is to be installed in. Magnetic fields are affected by absolutely everything so it is very important that the magnetometer calibration is done in the operational environment. Also bear in mind that many buildings severely distort the magnetic field inside.

timetogo2, 2015-06-05:
That worked for me.  Thank you so much for your help.  I also had another issue that was throwing me off sometimes and that is one of the locations where I was setup for testing was a desk with a formica top wrapped on metal underneath (steel-case desk).  This sometimes prevented me from even locating a zero yaw heading.  Putting my Galaxy S4 phone on any of these desks at work causes the compass application to severely misread and become worhtless as a compass.

I still see about 50x more variation/noise while sitting still on the yaw measurement than on the roll and pitch.  In your experience, are there better IMUs (with compass) than the 9150?  If I am only interested in relative motion (shaking, small sway, etc) on a fixed mounted platform could I set the COMPASS_ALPHA to a very small value?

Thanks again for your help,
Eric

richardbarnett, 2015-06-05:
Well we all learned from that. I have updated the RTIMULib readme to be a bit more specific about re-calibrating after changing the axis rotation setting.

Magnetic desks have caught me out in the past. Even a single magnetic screw or nail in a seemingly all wood desk can completely mess things up. I don't think anyone realizes how bad magnetic fields are until they try using them!

The STM chips (LSM9DS0 and L3GD20H/LSM303DLHC) seem to have much better magnetometers and can give adequate results even without calibration if there are no nearby disturbances. Certainly worth a try - there are many breakouts available for these chips.