richards-tech/RTIMULib #37: Missing version in SONAME

XECDesign, 2015-06-12:
I'm currently packaging up RTIMULib for inclusion in Raspbian and the Debian folk consider RTIMULib.so broken, since it doesn't have a proper SONAME.

More information here:
https://www.debian.org/doc/debian-policy/ch-sharedlibs.html
https://lintian.debian.org/tags/sharedobject-in-library-directory-missing-soname.html

Not a really an issue for our purposes, but thought it was worth reporting anyway.

richardbarnett, 2015-06-12:
Thanks for that. I'll take care of it.

richardbarnett, 2015-06-12:
There's now a V7.0.1 that should fix that for the CMake build but were you actually referring to the Python extension library?

XECDesign, 2015-06-13:
Thanks. I didn't get to the python part yet, so I'm not sure what will be needed there.

XECDesign, 2015-06-15:
Whoops, something is wrong.
```
~/dev/RTIMULib/RTIMULib/build $ readelf -d libRTIMULib.so | grep SONAME
 0x0000000e (SONAME)                     Library soname: [libRTIMULib.so]
```
Looking at the commit history, it seems like you've removed the version variables from RTIMULib/CMakeLists.txt in 6242504b08a9ec322e868594adf4cdeb10a9b4e1

richardbarnett, 2015-06-15:
I moved the version defs out to the main Linux CMakeLists.txt. I guess that's given you a problem. I'll put it back the way it was.

XECDesign, 2015-06-15:
No wait, I forgot that there was the main one. It's probably better for me to use that one. But if someone does build using the cmake files in rtimulib subdirectory, they'll get a slightly different  .so, which seems a little problematic.

richardbarnett, 2015-06-15:
I am certainly not a CMake expert so I wasn't sure what was the best thing to do here. It should be ok to just build the library and have it work the same. I just want to make sure that a consistent version number is applied (there is also an OpenGL lib - that's what caused the issue). I think that a better approach would be to include a file that contains the version info so that it works in all cases.

XECDesign, 2015-06-15:
Yeah, that sounds like a good approach.

On an unrelated side note. I saw that you were thinking about adding IIO support. If I managed to write IIO drivers for the devices we need, this was actually going to be my next PR. So it's definitely a +1 from me.

richardbarnett, 2015-06-15:
Ok. Let me know what I can do to help with that.

richardbarnett, 2015-06-15:
V7.0.3 should hopefully handle the CMake version correctly now!

XECDesign, 2015-06-15:
Seems okay, thank you.

closed 2015-06-15 by XECDesign