richards-tech/RTIMULib #39: Global/per-user RTIMULib.ini instead of per-directory

XECDesign, 2015-06-22:
It would be nice to have a ~/.config/RTIMULib.ini and /etc/RTIMULib.ini fallback instead of being relative to the working directory. Along the same lines, a sensible path for the ellipsoid fit stuff would be pretty good too.

richardbarnett, 2015-06-22:
There is a actually a second RTIMUSettings constructor that will accept a directory as well as a file name. Would that work for you?

Any thoughts on how to handle the ellipsoid situation? I agree - it's not a great situation.

XECDesign, 2015-06-22:
My mistake, forgot that it's up to the application to specify the location of the settings file. As long as there are no issues with using a full path with the ```RTIMUSettings::RTIMUSettings(const char *productType)``` constructor, I think we're fine.

As for the ellipsoid fit .m files, I'm not sure. What I was going to do was install them in /usr/share/librtimulib-utils/RTEllipsoidFit/ and patch RTIMULibCal to check there and use ~/.config/RTIMULib/ or /tmp for the .dta files, but this would be specific to this particular packaged version and not very upstreamable.

richardbarnett, 2015-06-23:
Yes, I think that you should be able to give a path instead of just the filename but, if not, the second constructor is available.

I will give some thought to the whole ellipsoid situation. It does need a more portable solution.

XECDesign, 2015-06-23:
Thanks again

closed 2015-06-23 by XECDesign