richards-tech/RTIMULib #40: Would it be possible to output raw and calibrated data together?

cyborg-x1, 2015-06-24:
I am currently working on adding calibration stuff to the i2c_imu ROS node ( https://github.com/jeskesen/i2c_imu ) and trying to create some autocalibration thing. So would it be possible to have both calibrated and raw data the same time? Or is there something transfered into the chip for the calibration. Like for example on the IMU9250, which has an internal motion processor ... ?

richardbarnett, 2015-06-25:
For what it's worth, the DMP on the MPU-9250 does not handle the magnetometers which are the things in most need of calibration.

Without modifying the library, there is no way to get both at the same time. It is possible to get either (separately) at any time though. Is that not enough? I am not clear why you'd need both simultaneously.

cyborg-x1, 2015-06-25:
For calibration reasons, I have a ground moving robot and thought of improving calibration by time. 
Actually I do not know yet if it would work out. Rotating the robot for the operator is not an option to get calibration data.

But for that approach I guess I'll need the raw data (accel and compass) all the time.  If you have that already it shouldn't be a problem, I probably could also have a look at it if that is the case ... We (you or me ...) could add another structure for the raw values. If I do it, it would be nice if you tell me in which file/function you get them.

richardbarnett, 2015-06-25:
If you look at line 308 in RTIMU.cpp you'll see the calibrateAverageCompass() function. m_imuData.compass is the uncalibrated value on entry. If you want a quick solution, create a new RTVector3 variable in RTIMU.h called (say) m_rawCompass and then you can use that for your calibration software. Add the line:

m_rawCompass = m_imuData.compass;

before line 310 for example.

Line 344 is the equivalent for accelerometer calibration.

cyborg-x1, 2015-06-25:
Added a function here, I just used RTVector3 for it, don't think its necessary to have all data twice.
https://github.com/Hacks4ROS-forks/RTIMULib/tree/feature_compass_raw